MajenkoLibraries/MCP23S17 #1: With 3 chips, the 3rd chip appears to function as an AND gate of the 1st two chips

dwilliams1994, 2018-09-04:
Here's my code, modified from the example code provided

`#include <MCP23S17.h>

#ifdef __PIC32MX__
// chipKIT uses the DSPI library instead of the SPI library as it's better
#include <DSPI.h>
DSPI0 SPI;
#else
// Everytying else uses the SPI library
#include <SPI.h>
#endif

const uint8_t chipSelect = 15 ;

// Create an object for each chip
// Bank 0 is address 0
// Bank 1 is address 1.
// Increase the addresses by 2 for each BA value.

MCP23S17 Bank1(new SPIClass(HSPI), chipSelect, 0);
MCP23S17 Bank2(new SPIClass(HSPI), chipSelect, 1);
MCP23S17 Bank3(new SPIClass(HSPI), chipSelect, 2);

void setup() {
  Serial.begin(115200);
  Bank1.begin();
  delay(50);
Bank2.begin();
delay(50);
  Bank3.begin();
  delay(50);
  for (int i = 0; i < 16; i++) {
    Bank1.pinMode(i, OUTPUT);
    delay(50);
   Bank2.pinMode(i, OUTPUT);
   delay(50);
    Bank3.pinMode(i, OUTPUT);
    delay(50);
  }
    for (int i = 0; i < 16; i++){
   Bank1.digitalWrite(i,  1);
   delay(50);
    Bank2.digitalWrite(i, 1);
    delay(50);
    Bank3.digitalWrite(i, 1);
    delay(50);
  }
}
void loop(){}`

The thing is, Bank3.digitalWrite(i, 1) does not appear to work. Instead, it follows and logic of the first 2. Ex: if Bank1.digitalWrite(i,1) and Bank2.digitalWrite(i,1) then Bank3's pins are pulled high. If either bank1 or bank2's pins are written low, then bank3's pins are pulled low, regardless of what Bank3's digitalWrite says.

I have triple checked my wiring and don't notice any issues

tonton81, 2018-09-04:
its not wiring its a code issue, actually an errata with the SPI chips, the first 4 (0,1,2,3) should work fine but 4,5,6,and 7 require a different opcode to initialize. did you tie all reset pins high? diagrams? please use code brackets when posting code, the errata may not apply to you, it may be something else

tonton81, 2018-09-04:
`void MCP23S17::begin() {
    _spi->begin();
    ::pinMode(_cs, OUTPUT);
    ::digitalWrite(_cs, HIGH);
    uint8_t cmd = 0b01000000;
    ::digitalWrite(_cs, LOW);
    _spi->transfer(cmd);
    _spi->transfer(IOCONA);
    _spi->transfer(0x18);
    ::digitalWrite(_cs, HIGH);
    writeAll();
}`

Issue looks to be here, the library is writing the first opcode but not the second. Also make sure your addresslines are properly configured via appropriate pullups/pulldowns

You then must apply both (0b01000000) OPCODE (in begin(), BUT, the begin should ALSO include OPCODE 0b01001110 (0x4F) to initialize chips 0 -> 3 for OPCODE 0x40 and 4-> for OPCODE 0x4E

Tony