MajenkoLibraries/MCP23S17 #2: Missing paragraph error in Arduino IDE

addicore, 2020-05-27:
While loading the library into the Arduino IDE, I am getting a missing paragraph error. I was able to track it down to the library.properties file missing a paragraph and url line.

`paragraph=`
`url=https://github.com/MajenkoLibraries/MCP23S17.git`

Adding those 2 lines to the library.properties seems to resolve the issue.

This error happens in the current version (4/25) and 4/17 commits.

majenkotech, 2020-05-27:
Strange, I would have expected paragraph to be an optional field, since it is merely an extension of what is in sentence. Ft00pid Arduino...

These library.properties files are all generated by a script, so I've added that to the script and I'm re-running it for all my libraries.

majenkotech, 2020-05-27:
Should be fixed in f43cdc5

addicore, 2020-05-27:
We'll check it out, thanks!

Andre

closed 2021-08-09 by majenkotech