MajenkoLibraries/MCP23S17 #4: Ravi N

nravirke, 2021-01-11:
HI,
Thanks for the Great Work.

Can I Connect 10 Numbers of MCP23S17

I need to know the limitation of the number of MCP23S17 can be connected..

Thanks in advance.

majenkotech, 2021-01-11:
Theoretically, with 3 address lines, you can connect 2^3, or 8, chips to a single CS pin. Add a second CS pin from your microcontroller and you have 2 buses of up to 8 devices - that is, 16 devices. A third CS pin gives you 8 more, so you're up to 24 devices. 

The biggest issue is the "fan-out" limit of the SPI pins for your microcontroller - that is, the maximum capacitance that can exist connected to an output pin before it starts to get overloaded and can't switch fast enough. For very large numbers of chips you may need to add extra buffers to drive them in sub-groups, or reduce the clock speed to allow more time for switching to take place.

nravirke, 2021-01-11:
Thanks for a quick reply.

nravirke, 2021-04-15:
Hi,
This is Ravi again seeking your help.
Have you done any Library for MCP23S17 (Similar to ATMEGA 328P) For PIC Micro.
Thanks & Regards,

majenkotech, 2021-04-15:
For what PIC micros? There's a whole range of chip types under the "PIC" umbrella.

nravirke, 2021-04-15:
For PIC 18F4550 ..

majenkotech, 2021-04-15:
No. There's no "standard" library system for that like there is for Arduino. I usually work direct with the SPI / I2C registers directly for those kind of chips. With minimal resources writing lean code is more important than importing chunks from elsewhere.

nravirke, 2021-04-15:
Thanks for the Reply.
Is it possible to develop / Port from Arduino to PIC 18F4550..

majenkotech, 2021-04-15:
Not as such, no. The lack of a C++ compiler means that you don't get any of the OOP that Arduino relies on. It's possible to kind of get the feel of it using structs, but it's very limited and doesn't work at all well. 

I did at one point make an experimental Arduino-like framework for the low-end PICs but it really was hacky and only Ardiuno-like in the barest sense.

* https://github.com/UECIDE-Cores/pic18

nravirke, 2021-12-10:
Hi,
Thanks for your email.
I built the Circuit & it is Working. But I need to know how to alter the
SPI bus speed.
Thanks & Regards,
Ravi

majenkotech, 2021-12-10:
You would want to add a line after [this one](https://github.com/MajenkoLibraries/MCP23S17/blob/master/src/MCP23S17.cpp#L123) with the new clock settings in it.  Such as:

    _spi->setClockDivider(SPI_CLOCK_DIV16);

nravirke, 2021-12-10:
Hi,
What a fast Reply sir.
Thanks a Lot.
I will try.
Thanks once again.

nravirke, 2021-12-17:
Hi,
This is Ravi from Bengaluru, India. As suggested by you I could change the
SPI Speed down to 1MHz.
I am using your Library for MCP23s17. Here I am sending a Program file for
the Control of 128 Relays.
I am using a different power supply for relays & Atmega328P ( I have a
standalone Atmega328 Board).
It works fine, but sometimes it stops working & sometimes it requires reset
after switching on Power Supply.
Consistency is not there. Request go through the code & Have your Comment.
The length of cable from controller to MCP23s17 PCB is about .75mts.
Pl suggest any Improvement of the same.

Thanks in advance for your time.

with regards,

Ravi

majenkotech, 2021-12-18:
There can be many causes of random crashes: over-reliance on the `String` object, EMI, poor quality or underpowered power supply, etc. You can work around many problems by using the watchdog timer to reset the board for you when it locks up.

nravirke, 2022-02-02:
Dear Sir,
Pl confirm if it  Possible to Write a Library for MCP23S17 to Work On PIC
18F4550.

It should be Similar to the Library written for ATmega328P.

If so Pl Confirm. & Let Me know what will be the Approximate Development
Charges for the same.

With regards,

Ravi

nravirke, 2023-05-15:
Dear Sir,
This is Ravi from India.
I am working on the STM32f103 series.. with STM32 cube IDE.
I need to control MCP23S17.  I Had gone through your MCP23S17 lIBRARAY For
the Arduino.
Is it possible to port the Library to use under CUBE IDE.
Please confirm.
Thanks in advance.

with regards,
Ravi

majenkotech, 2023-05-15:
I have no idea. I've never heard of the CUBE IDE. I can't help you.

closed 2023-05-15 by majenkotech