MajenkoLibraries/MCP23S17 #5: Daisy-chaining support?

OmerFlame, 2021-02-28:
Is that supported on your library? Also, in your example you wrote: `Increase the addresses by 2 for each BA value.`. Can you explain what that means?

Thank you!

majenkotech, 2021-02-28:
Multiple chips are supported, though it isn't "daisy chaining" like you do with shift registers. Each chip is a separate entity on a multi-drop bus.

The example is written for a specific product that has two chips on it with subsequent chip addresses (A0 set to 0 on one and 1 on the other) so multiple boards will have multiples-of-two addresses.

OmerFlame, 2021-02-28:
Ah, got it. Is shift-register daisy-chaining supported by this chip at all?

majenkotech, 2021-02-28:
No. It's not a shift register. It's an SPI slave device with internal registers and such - a shift register is a dumb device that just outputs whatever you send it or reads whatever is on the pins.

OmerFlame, 2021-02-28:
Thank you for the very quick response and helpful feedback! I will now close the issue.

closed 2021-02-28 by OmerFlame