MajenkoLibraries/MCP23S17 #7: Only first digitalWrite takes effect

pajasry, 2021-04-24:
Hi, I am using ESP32 and MCP23S17 with this library, but I am experiencing weird behavior. When I try simple blink program, only first digitalWrite is applied.

In this case pin 6(GPA6) i always on, but when I switch digitalWrite statements(LOW first, then HIGH) pin persists LOW
```c++
#include <Arduino.h>
#include <SPI.h>
#include <MCP23S17.h> 

SPIClass spi(HSPI);
MCP23S17 mcp23s17( &spi, 15, 0x20 );

void setup()
{
  Serial.begin(115200);
  mcp23s17.begin();
  mcp23s17.setInterruptLevel(HIGH);
  mcp23s17.pinMode(6,OUTPUT);

  
  Serial.println("ON");
}

void loop(){
  mcp23s17.digitalWrite(6,HIGH);
  Serial.println("LOW");
  delay(1000);
  mcp23s17.digitalWrite(6,HIGH);
  Serial.println("HIGH");
  delay(1000);
}

```
Any advice? I am missing something very basic here, but I can't figure it.

majenkotech, 2021-04-25:
I have never tried this library with the ESP32. I will have to dig some hardware out and set up a test environment to debug it. I take it you're seeing all the `Serial.println()` commands fine?

majenkotech, 2021-04-25:
I have just rigged up a test, and with your code verbatim (except of course setting the first HIGH to LOW) I get a blinking LED.

One thing: the address of 0x20 is not right. There's only a 3-bit address, so it has to be between 0 and 7. The number you send is masked with 0b111, so 0x20 becomes 0x00 anyway. Just FYI...

So... can you post a picture of your setup? Maybe there's something amiss with your wiring.

pajasry, 2021-04-26:
I have QFN package, connected like this:
[image]

Now I'm thinking unconnected A0-A3 can be a problem?

majenkotech, 2021-04-26:
Yes, certainly. You can't leave them floating, they need to be connected to GND for address 0.

pajasry, 2021-04-26:
I pinned them to GND and everything works like a charm. Thank you for quick responses, patient and good Library 👍

closed 2021-04-26 by pajasry