MajenkoLibraries/MCP23S17 #10: Compile errors

thijstriemstra, 2021-08-09:
The echo example included in this library throws the following errors with platformio:

```
In file included from src/main.cpp:6:0:
.pio/libdeps/mcp23017_test/MCP23S17/src/MCP23S17.h:83:27: error: expected ')' before '*' token
         MCP23S17(SPIClass *spi, uint8_t cs, uint8_t addr);
                           ^
.pio/libdeps/mcp23017_test/MCP23S17/src/MCP23S17.h:84:27: error: expected ')' before '&' token
         MCP23S17(SPIClass &spi, uint8_t cs, uint8_t addr);
                           ^
.pio/libdeps/mcp23017_test/MCP23S17/src/MCP23S17.h:47:16: error: an anonymous struct cannot have function members
 class MCP23S17 {
                ^
.pio/libdeps/mcp23017_test/MCP23S17/src/MCP23S17.h:103:1: error: abstract declarator '<anonymous class>' used as declaration
 };
 ^
src/main.cpp:27:15: error: expected constructor, destructor, or type conversion before '(' token
 MCP23S17 Bank1(&SPI, chipSelect, 0);
               ^
src/main.cpp:28:15: error: expected constructor, destructor, or type conversion before '(' token
 MCP23S17 Bank2(&SPI, chipSelect, 1);
               ^
*** [.pio/build/mcp23017_test/src/main.cpp.o] Error 1
```

majenkotech, 2021-08-09:
What target are you trying to compile for?

thijstriemstra, 2021-08-09:
this was for esp32 wrover-kit in platformio

majenkotech, 2021-08-09:
The the ESP32 core probably doesn't have an SPIClass for its SPI system. This library is written first and foremost for Arduino and chipKIT. If you'd like to port it to other targets then feel free and make a PR with the needed changes.

thijstriemstra, 2021-08-09:
It does have an SPIClass: https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h

I have a feeling these errors are more about a stricter syntax that platformio requires vs arduino IDE. Meaning these errors will also show when compiling for AVR with platformio.

majenkotech, 2021-08-09:
You may be right - it compiles perfectly for esp32:esp32:lolin32 with arduino-cli. Does platformIO know to include SPI.h in the compilation?

thijstriemstra, 2021-08-09:
>  Does platformIO know to include SPI.h in the compilation?

Yea, never saw any errors there..

majenkotech, 2021-08-09:
Well, it's acting like it doesn't know what SPIClass is. If it's including the right SPI.h file then it should be finding it, but isn't. Do you have another SPI.h file in your project that's confusing it?

thijstriemstra, 2021-08-09:
platformio and d1 mini seem to work fine:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (3.1.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.30001.210627 (3.0.1) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MCP23S17> 1.1.3+sha.2bfee60
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
...
Linking .pio/build/mcp23017_test/firmware.elf
Retrieving maximum program size .pio/build/mcp23017_test/firmware.elf
Checking size .pio/build/mcp23017_test/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.3% (used 28068 bytes from 81920 bytes)
Flash: [===       ]  25.2% (used 262841 bytes from 1044464 bytes)
Building .pio/build/mcp23017_test/firmware.bin
```

majenkotech, 2021-08-09:
Can you paste the full build log for the esp32?

thijstriemstra, 2021-08-09:
Not sure why my initial setup wasn't working but it's now also running without compile errors in platformio using esp-wrover-kit. Sorry for the noise!

closed 2021-08-09 by thijstriemstra

majenkotech, 2021-08-09:
It must be a monday... ;)