MajenkoLibraries/MCP23S17 #11: Interrupt example

thijstriemstra, 2021-08-09:
It would be nice to add an example that shows how to use the interrupts with one or more buttons (and leds).

@majenkotech if you can give me some pointers I can make a pull request for the example if you want.

majenkotech, 2021-08-09:
Interrupts are a case of configuring the chip to indicate an interrupt and wiring the right INT pin to an interrupt pin on your MCU. I can't recall what support I have coded into the library for that of hand, I'll have to dig into it.

majenkotech, 2021-08-09:
Everything you need should be in the manual. 

https://github.com/MajenkoLibraries/MCP23S17/blob/master/latex/refman.pdf

thijstriemstra, 2021-08-09:
thanks I'll take a look. Also opened #12 for browsable online html docs.

thijstriemstra, 2021-08-09:
Not much luck with that PDF yet, but http://84park.blogspot.com/2016/12/arduino-and-mcp23s17-port-expander-part_29.html contains quite some useful info.

majenkotech, 2021-08-09:
It should be a case of calling `.enableInterrupt(...)` to turn it on in the chip, then in your ISR you'd just call `.getInterruptPins()` to find what triggered it. You can call `.setMirror(true)` to make both interrupt pins act as one.

majenkotech, 2021-08-09:
That link is very interesting though. I'm integrating their changes into the library.

majenkotech, 2021-08-09:
Ok, changes integrated.

thijstriemstra, 2021-08-09:
> It should be a case of calling .enableInterrupt(...) to turn it on in the chip, then in your ISR you'd just call .getInterruptPins() to find what triggered it. You can call .setMirror(true) to make both interrupt pins act as one.

I'll try to create a minimal working example.

Here's the (modified) example from http://84park.blogspot.com/2016/12/arduino-and-mcp23s17-port-expander-part_29.html that seems to work well but (probably) has a lot of unneeded code. Tested with Arduino Nano v3 clone.

```cpp
#include <Arduino.h>
#include <MCP23S17.h>
#include <SPI.h>

const uint8_t address = 0;
const uint8_t interruptPinA = 2;
const uint8_t chipSelectPin = 10;

volatile uint16_t mcpReading = 0; // 16-bit interrupt reading
volatile uint8_t mcpAReading = 0; // 8-bit Port-A interrupt reading
volatile uint8_t mcpBReading = 0; // 8-bit Port-B interrupt reading
volatile uint8_t flagMCPA = 0;
volatile uint8_t flagINTA = 0;
volatile uint8_t flagMCPB = 0;
volatile uint8_t flagINTB = 0;
uint8_t whichBit = 0;    // one bit [7...0] that is set in byte
uint8_t byteACount = 0; // number of bits set in byte
uint8_t byteBCount = 0; // number of bits set in byte
uint8_t byteABits[8] = {0}; // set bit array
uint8_t byteBBits[8] = {0}; // set bit array

MCP23S17 mcp23s17(&SPI, chipSelectPin, address);

void print8Bits(uint8_t myByte) {
  for (uint8_t mask = 0x80; mask; mask >>= 1) {
    if (mask  & myByte)
      Serial.print('1');
    else
      Serial.print('0');
  }
}

//---------------------------------------------------------------------------------
// print 16-bit word as 16 bit binary string
//---------------------------------------------------------------------------------

void print16Bits(uint16_t myWord) {
  for (uint16_t mask = 0x8000; mask; mask >>= 1) {
    if (mask & myWord)
      Serial.print('1');
    else
      Serial.print('0');
  }
}

//---------------------------------------------------------------------------------
// crPrintHEX print value as hex with specified number of digits
//---------------------------------------------------------------------------------

void crPrintHEX(unsigned long DATA, unsigned char numChars) {
  unsigned long mask  = 0x0000000F;
  mask = mask << 4 * (numChars - 1);
  Serial.print("0x");
  for (unsigned int eID = numChars; eID > 0;  --eID) {
    Serial.print(((DATA & mask) >> (eID - 1) * 4), HEX);
    mask = mask >> 4;
  }
  Serial.print("  ");
}

// Set all Port Expander pins to desired mode INPUT_PULLUP
// Set all Port Expander input pins interrupts
void setPin(MCP23S17 &bank) {
[... 170 lines cut ...]
  mcpBReading = 0;
  mcpReading = 0;
}

void loop() {
  // This code runs the MCP Interrupt processing from the Arduino ISR
  // and then takes action (prints) on the MCP pins that went LOW

  if (flagMCPA > 0) {
    //Serial.println("loop flagMCPA");
    printInterruptHeader(mcp23s17, 0);
    processPins(mcp23s17, 0); // do some processing of each interrupted pin
    flagMCPA = 0;
  }

  if (flagMCPB > 0) {
    //Serial.println("loop flagMCPB");
    printInterruptHeader(mcp23s17, 1);
    processPins(mcp23s17, 1); // do some processing of each interrupted pin
    flagMCPB = 0;
  }

  //delay(10); // give loop something to do while idle
}
```

majenkotech, 2021-08-09:
Great. Added it as an example.

thijstriemstra, 2021-08-09:
> You can call .setMirror(true) to make both interrupt pins act as one.

@majenkotech do I have to physically tie INTA and INTB together? how, any resistors needed?

majenkotech, 2021-08-09:
No. This makes both respond in the same way as if they were both tied together inside the chip.

closed 2021-08-09 by thijstriemstra

thijstriemstra, 2021-08-09:
I wasn't sure how to update the Interrupt example to use a single Arduino interrupt for both port A and B so I ended up using 2 interrupt pins. According to the [article](http://84park.blogspot.com/2016/12/arduino-and-mcp23s17-port-expander-part_29.html):

>  What does one do if they want to use interrupts for more than one MCP23S17 chip? Where do the MCP23S17 INTA and INTB pins connect on the Arduino? The simple answer is to use a processor, like the Mega 2560, with more interrupts. 

So I'll use 2 interrupt pins on my ESP32 for each MCP23S17 (2 in total) = 4 ESP32 pins for 32 buttons. Not bad.

majenkotech, 2021-08-09:
It depends how the interrupt pin is configured.  If you set it to "Open Drain" mode then add a pullup all the interrupt pins can wire to the same Arduino INT pin. But then you have to cycle through all the chips to find out which one caused the interrupt.